rednblkx/HAP-ESPHome #3: Installing on an ESP32-WROOM-32 (AZ-Delivery) fails

pietro4711, 2024-08-04:
Hi,

probably I am too stupid - but firmware uploading fails with following message:

`/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: .pioenvs/esphome-web-e9a788/firmware.elf section `.iram0.text' will not fit in region `iram0_0_seg'
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: IRAM0 segment data does not fit.
/data/cache/platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: region `iram0_0_seg' overflowed by 4616 bytes` 

This is my yaml file:

`substitutions:
  name: esphome-web-e9a788
  friendly_name: HomeKey

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.6.0
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: '1.0'

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    version: 5.2.1
    platform_version: 6.7.0
    sdkconfig_options:
      CONFIG_COMPILER_OPTIMIZATION_SIZE: y
      CONFIG_LWIP_MAX_SOCKETS: "16"
      CONFIG_MBEDTLS_HKDF_C: y

external_components:
  source: github://rednblkx/HAP-ESPHome@main

# Enable logging
logger:

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
- platform: esphome

# Allow provisioning Wi-Fi via serial
improv_serial:

wifi:
  # Set up a wifi access point
  ap: {}

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

dashboard_import:
  package_import_url: github://esphome/example-configs/esphome-web/esp32.yaml@main
  import_full_config: true

# Sets up Bluetooth LE (Only on ESP32) to allow the user
# to provision wifi credentials to the device.
esp32_improv:
  authorizer: none

# To have a "next url" for improv serial
web_server:

lock:
  - platform: template
    id: "this_lock"
    name: "Main Lock"
    optimistic: True
    on_lock:
    - logger.log: "Door Locked!"
    on_unlock:
    - logger.log: "Door Unlocked!"

homekit_base:
  setup_code: '159-35-728'

homekit:
  lock:
    - id: this_lock
      nfc_id: nfc_spi_module
      on_hk_success:
        lambda: |-
          ESP_LOGI("HEREHERE", "IssuerID: %s", x.c_str());
          ESP_LOGI("HEREHERE", "EndpointID: %s", y.c_str());
      on_hk_fail:
        lambda: |-
          ESP_LOGI("GSDGSGS", "IT FAILED :(");
      hk_hw_finish: "BLACK"

spi:
  clk_pin: 18
  miso_pin: 19
  mosi_pin: 23

pn532_spi:
  id: nfc_spi_module
  cs_pin: 5
  update_interval: 100ms`

Please help!

pietro4711, 2024-08-04:
just removed some unnecessary stuff like bluetooth and web server -> now it fits into the memory :-)

closed 2024-08-04 by pietro4711

rednblkx, 2024-08-04:
Yeah, Bluetooth takes up a lot of RAM space, you should be alright with everything else