rednblkx/HAP-ESPHome #4: cant find Attributes IssuerID or EndpointID

pietro4711, 2024-08-04:
Here is my config:

`homekit:
  lock:
    - id: this_lock
      nfc_id: nfc_spi_module
      on_hk_success:
        lambda: |-
          ESP_LOGI("HEREHERE", "IssuerID: %s", x.c_str());
          ESP_LOGI("HEREHERE", "EndpointID: %s", y.c_str());
      on_hk_fail:
        lambda: |-
          ESP_LOGI("GSDGSGS", "IT FAILED :(");
      hk_hw_finish: "SILVER"

`
But I cant find any attributes after a successful iOS Authentication.

rednblkx, 2024-08-04:
Note that those are only parameters of type `std::string` passed on as variable `x` and `y` to `on_hk_success:` to be used in a lambda function like in your case, and in your config those are just printed to the terminal.

There is no integration with HA like in HomeKey-ESP32, at least not at this moment.

pietro4711, 2024-08-04:
Ok, thanks!
So for the moment if I want some HA automations depending on the user ID, Homekey-ESP32 is still the way to go, right?

rednblkx, 2024-08-04:
Well, yeah, for the moment, if you just want something straightforward HomeKey-ESP32 is the option.

But like I said, the IDs are exposed on `on_hk_success` so anyone that knows how to make the use of it, should be able to use them.

closed 2024-08-04 by rednblkx