rednblkx/HAP-ESPHome #6: Compile Error

adrum, 2024-08-05:
I'm getting some compile issues. I've tried a variety of different things. This is my most recent config:

```yaml
esphome:
  name: front-door-homekey
  friendly_name: Front Door HomeKey

external_components:
  # use rtttl and dfplayer from ESPHome's dev branch in GitHub
  - source:
      type: git
      url: https://github.com/rednblkx/HAP-ESPHome
      ref: main
    components: [ homekit, homekit_base, pn532, pn532_spi ]

# Enable Home Assistant API
api:

ota:
  - platform: esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

lock:
  - platform: template
    id: "front_door_homekey"
    name: "Froont Door Lock HomeKey"
    optimistic: True
    on_lock:
      # - logger.log: "Door Locked!"
      - homeassistant.service:
          service: lock.lock
          data:
            entity_id: lock.front_door_lock
    on_unlock:
      - homeassistant.service:
          service: lock.unlock
          data:
            entity_id: lock.front_door_lock
      # - logger.log: "Door Unlocked!"

sensor:
  # - platform: template
  #   id: "my_sensor"
  #   name: "Template Sensor"
  #   unit_of_measurement: "°C"
  #   icon: "mdi:water-percent"
  #   device_class: "temperature"
  #   state_class: "measurement"
  #   lambda: |-
  #       return id(number_sensor).state;
  #   update_interval: 10s
  # - platform: template
  #   id: "my_sensor_second"
  #   name: "Template Sensor 1"
  #   unit_of_measurement: "°C"
  #   icon: "mdi:water-percent"
  #   device_class: "temperature"
  #   state_class: "measurement"
  #   lambda: |-
  #       return id(number_sensor).state;
  #   update_interval: 10s

number:
  - platform: template
    name: "Sensor Value"
    min_value: 10
    max_value: 30
    step: 1
    optimistic: True
    id: "number_sensor"

switch:
  # - platform: template
  #   id: some_switch
  #   name: "Template Switch"
  #   restore_mode: RESTORE_DEFAULT_OFF
  #   lambda: |-
  #     if (id(test_light).current_values.get_state()) {
  #       return true;
  #     } else {
  #       return false;
  #     }
  #   turn_on_action:
  #     - light.turn_on: test_light
  #   turn_off_action:
  #     - light.turn_off: test_light

spi:
  clk_pin: 18
  miso_pin: 19
  mosi_pin: 23

pn532_spi:
  id: nfc_spi_module
  cs_pin: 5
  update_interval: 100ms

esp32:
  # board: esp32-c3-devkitm-1
  board: esp32dev
  flash_size: 4MB
  framework:
    type: esp-idf
    version: 5.2.1
    platform_version: 6.7.0
    sdkconfig_options:
      CONFIG_COMPILER_OPTIMIZATION_SIZE: y
      CONFIG_LWIP_MAX_SOCKETS: "16"
      CONFIG_LOG_DEFAULT_LEVEL_ERROR: y
      CONFIG_LOG_DEFAULT_LEVEL: "1"
      CONFIG_LOG_MAXIMUM_LEVEL_INFO: y
      CONFIG_LOG_MAXIMUM_LEVEL: "3"
      CONFIG_MBEDTLS_HKDF_C: y

homekit_base:
  setup_code: '159-35-728'

homekit:
  lock:
    - id: front_door_homekey
      nfc_id: nfc_spi_module
      # on_hk_success:
        # lambda: |-
        #   ESP_LOGI("HEREHERE", "IssuerID: %s", x.c_str());
        #   ESP_LOGI("HEREHERE", "EndpointID: %s", y.c_str());
        # lambda: |-
        #   ESP_LOGI("HEREHERE", "IssuerID: %s", x.c_str());
        #   ESP_LOGI("HEREHERE", "EndpointID: %s", y.c_str());
        #   id(test_light).toggle().perform();
      # on_hk_fail:
      #   lambda: |-
      #     ESP_LOGI("GSDGSGS", "IT FAILED :(");
      hk_hw_finish: "SILVER"
  light:
    # - id: test_light
    # - id: desk_light
  sensor:
    # - id: my_sensor
    # - id: my_sensor_second
  switch:
    # - id: some_switch

button:
  - platform: homekit_base
    factory_reset:
      name: "Reset Homekit pairings"

light:
  # - platform: esp32_rmt_led_strip
  #   id: test_light
  #   rgb_order: GRB
  #   pin: GPIO8
  #   num_leds: 1
  #   rmt_channel: 0
  #   chipset: ws2812
  #   name: "My Light"
  #   restore_mode: RESTORE_DEFAULT_OFF

  # - platform: binary
  #   id: desk_light
  #   name: "Desk Lamp"
  #   output: simple_led
  #   restore_mode: RESTORE_DEFAULT_OFF

logger:
  # hardware_uart: UART0
  level: DEBUG
```

Here's my output:

```
INFO ESPHome 2024.7.3
INFO Reading configuration /config/esphome/front-door-nfc.yaml...
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.2.5
INFO Updating https://github.com/espressif/idf-extra-components.git@master
INFO Updating submodules (libsodium/libsodium) for https://github.com/espressif/idf-extra-components.git@master
INFO Updating https://github.com/rednblkx/esp-homekit-sdk@master
INFO Updating https://github.com/rednblkx/HK-HomeKit-Lib.git@esp-idf
INFO Compiling app...
Processing front-door-homekey (board: esp32dev; framework: espidf; platform: platformio/espressif32@6.7.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50201.240515 (5.2.1) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp-elf @ 13.2.0+20230928
Reading CMake configuration...
Warning! Flash memory size mismatch detected. Expected 4MB, found 2MB!
Please select a proper value in your `sdkconfig.defaults` or via the `menuconfig` target!
Dependency Graph
|-- noise-c @ 0.1.4
Compiling .pioenvs/front-door-homekey/lib0e0/libsodium/crypto_aead/chacha20poly1305/sodium/aead_chacha20poly1305.o
Compiling .pioenvs/front-door-homekey/lib0e0/libsodium/crypto_core/ed25519/core_ed25519.o
Compiling .pioenvs/front-door-homekey/lib0e0/libsodium/crypto_core/ed25519/core_ristretto255.o
Compiling .pioenvs/front-door-homekey/lib0e0/libsodium/crypto_core/ed25519/ref10/ed25519_ref10.o
In file included from .piolibdeps/front-door-homekey/libsodium/libsodium/src/libsodium/crypto_aead/chacha20poly1305/sodium/aead_chacha20poly1305.c:13:
components/HK-HomeKit-Lib/include/utils.h:2:10: fatal error: string: No such file or directory
    2 | #include <string>
      |          ^~~~~~~~
compilation terminated.
*** [.pioenvs/front-door-homekey/lib0e0/libsodium/crypto_aead/chacha20poly1305/sodium/aead_chacha20poly1305.o] Error 1
In file included from .piolibdeps/front-door-homekey/libsodium/libsodium/src/libsodium/include/sodium/private/ed25519_ref10_fe_25_5.h:4,
                 from .piolibdeps/front-door-homekey/libsodium/libsodium/src/libsodium/include/sodium/private/ed25519_ref10.h:25,
                 from .piolibdeps/front-door-homekey/libsodium/libsodium/src/libsodium/crypto_core/ed25519/core_ed25519.c:6:
components/HK-HomeKit-Lib/include/utils.h:2:10: fatal error: string: No such file or directory
    2 | #include <string>
      |          ^~~~~~~~
compilation terminated.
*** [.pioenvs/front-door-homekey/lib0e0/libsodium/crypto_core/ed25519/core_ed25519.o] Error 1
In file included from .piolibdeps/front-door-homekey/libsodium/libsodium/src/libsodium/include/sodium/private/ed25519_ref10_fe_25_5.h:4,
                 from .piolibdeps/front-door-homekey/libsodium/libsodium/src/libsodium/include/sodium/private/ed25519_ref10.h:25,
                 from .piolibdeps/front-door-homekey/libsodium/libsodium/src/libsodium/crypto_core/ed25519/core_ristretto255.c:7:
components/HK-HomeKit-Lib/include/utils.h:2:10: fatal error: string: No such file or directory
    2 | #include <string>
      |          ^~~~~~~~
compilation terminated.
*** [.pioenvs/front-door-homekey/lib0e0/libsodium/crypto_core/ed25519/core_ristretto255.o] Error 1
In file included from .piolibdeps/front-door-homekey/libsodium/libsodium/src/libsodium/include/sodium/private/ed25519_ref10_fe_25_5.h:4,
                 from .piolibdeps/front-door-homekey/libsodium/libsodium/src/libsodium/include/sodium/private/ed25519_ref10.h:25,
                 from .piolibdeps/front-door-homekey/libsodium/libsodium/src/libsodium/crypto_core/ed25519/ref10/ed25519_ref10.c:8:
components/HK-HomeKit-Lib/include/utils.h:2:10: fatal error: string: No such file or directory
    2 | #include <string>
      |          ^~~~~~~~
compilation terminated.
*** [.pioenvs/front-door-homekey/lib0e0/libsodium/crypto_core/ed25519/ref10/ed25519_ref10.o] Error 1
========================= [FAILED] Took 24.39 seconds =========================
```

sven253, 2024-08-05:
no professional here but looks like you have to less memory:
`Warning! Flash memory size mismatch detected. Expected 4MB, found 2MB!`

rednblkx, 2024-08-05:
yeah, that's the same issue as #2, using `api:` without `encryption:` should solve it, i tested it and works, i see you don't have the `encryption:` property so i assume something got left behind, try deleting the `.esphome` folder and run again

Essentially, both the crypto library `libsodium` and `HK-HomeKit-Lib` used for homekey things have this file `utils.h` and the file from the homekey lib ends up being used in both.

It will be fixed soon.

the-smart-home-maker, 2024-08-07:
removing `encryption:` from `api:` works and solves the problem. Thank you!

rednblkx, 2024-08-07:
fixed, `encryption:` now works and above error should no longer occur

adrum, 2024-08-07:
Verified! I had the `encryption:` option in my config, but I deleted when pasting into here. My apologies for the confusion, and thank you for fixing the issue which allows for the encryption option to be enabled 🙌

closed 2024-08-07 by adrum