rednblkx/HAP-ESPHome #8: Unable to Add Accessory - This accessory cannot be used with HomeKit

aferrato, 2024-08-06:
Hi, when I try to add the accessory to Home, I receive this error: Unable to Add Accessory - This accessory cannot be used with HomeKit

Below are the logs.
Thanks.

```
[16:28:50][D][esp-idf:000][httpd]: I (29691) HAP: ######## Starting Pair Verify ########
[16:28:50]
[16:28:50]
[16:28:50][D][esp-idf:000][httpd]: I (29692) HAP: Pair Verify M1 Received
[16:28:50]
[16:28:50]
[16:28:50][D][esp-idf:000][httpd]: I (29762) HAP: Pair Verify M2 Successful
[16:28:50]
[16:28:50]
[16:28:50][W][component:237]: Component time took a long time for an operation (64 ms).
[16:28:50][W][component:238]: Components should block for at most 30 ms.
[16:28:50][D][esp-idf:000][httpd]: I (29866) HAP: Pair Verify M3 Received
[16:28:50]
[16:28:50]
[16:28:50][D][esp-idf:000][httpd]: I (29900) HAP: HomeKit Session active
[16:28:50]
[16:28:50]
[16:28:50][D][esp-idf:000][httpd]: I (29901) HAP: Pair Verify Successful for 441C9CB4-6940-4051-89DC-A7C6C6FE1B6F
[16:28:50]
[16:28:50]
[16:28:50][D][esp-idf:000][httpd]: I (30139) HAP: Decryption error/Connection lost. Marking session as invalid
[16:28:50]
[16:28:50]
[16:28:51][D][esp-idf:000][httpd]: I (30144) HAP: HomeKit Session terminated
[16:28:51]
[16:28:51]
[16:28:52][D][esp-idf:000][httpd]: I (31966) HAP: ######## Starting Pair Verify ########
[16:28:52]
[16:28:52]
[16:28:52][D][esp-idf:000][httpd]: I (31967) HAP: Pair Verify M1 Received
[16:28:52]
[16:28:52]
[16:28:52][D][esp-idf:000][httpd]: I (32035) HAP: Pair Verify M2 Successful
```

rednblkx, 2024-08-06:
yeah it's a very rare issue that occurs on pairing and hard to pinpoint the cause due to its rarity.

Use the factory reset button or just erase the whole thing and flash again.

aferrato, 2024-08-07:
After erasing the ESP32 and reflashing it, it works! Thank you.

closed 2024-08-07 by aferrato