rednblkx/HAP-ESPHome #9: Feature Request: Customizable Accessory Information

adrum, 2024-08-08:
It would be nice to be able to customize the Accessory Information attributes for both the Bridge Accessory, and the Child Accessories, such as:

- Manufacturer
- Serial Number
- Model
- Firmware Version

adrum, 2024-08-08:
Additionally, it might be nice to be able to customize the default bridge accessory name as well.

rednblkx, 2024-08-08:
I'll keep that mind, thanks

rednblkx, 2024-08-27:
The accessory information can now be modified for both the bridge and each individual accessory, see [documentation](https://github.com/rednblkx/HAP-ESPHome?tab=readme-ov-file#3-components)

I'll be closing this now, let me know if anything else, you can join the discord server if you wish https://discordapp.com/invite/VWpZ5YyUcm

closed 2024-08-27 by rednblkx