rednblkx/HAP-ESPHome #13: Error, Can't decode message length.

the-smart-home-maker, 2024-08-18:
Hi there,

I have everything set up now. Adding the bridge and the lock to the Apple Home App works like a charm. But then, when I want to unlock the lock with my Apple Watch or iPhone, the function `on_hk_success` will be successfully called but I get the following error

```
[E][nfc:058]: Error, Can't decode message length.
```

and the corresponding template lock does not lock / unlock. Maybe I understand it wrongly but if have my YAML code like the following, I would expect that the template lock with the name cupboard_lock does automatically lock / unlock, shouldn't it?

```
spi:
  clk_pin: GPIO18
  miso_pin: GPIO19
  mosi_pin: GPIO23

pn532_spi:
  id: nfc_spi_module
  cs_pin: GPIO26
  update_interval: 100ms

homekit_base:
  setup_code: '466-37-726'

homekit:
  lock:
    - id: cupboard_lock
      nfc_id: nfc_spi_module
      on_hk_success:
        lambda: |-
          ESP_LOGI("TEST", "IssuerID: %s", x.c_str());
          ESP_LOGI("TEST", "EndpointID: %s", y.c_str());
      on_hk_fail:
        lambda: |-
          ESP_LOGI("TEST", "IT FAILED");
      hk_hw_finish: "SILVER"

button:
- platform: homekit_base
  factory_reset:
    name: "Reset Homekit pairings"

lock:
  - platform: template
    id: cupboard_lock
    name: "HomeKit Cupboard Lock"
    optimistic: True
    on_lock:
    - logger.log: "Door Locked!"
    - switch.turn_off: electronic_lock
    on_unlock:
    - logger.log: "Door Unlocked!"
    - switch.turn_on: electronic_lock
    - delay:  100ms
    - switch.turn_off: electronic_lock

output:
  - platform: gpio
    pin: GPIO25
    id: 'lock_out'
switch:
  - platform: output
    id: electronic_lock
    output: 'lock_out'
```

This is the log output including the error message mentioned above but as you see the `cupboard_lock` template lock will NOT be locked / unlocked (no log output "Door Unlocked!" / "Door Locked!".

```
[08:08:10][D][pn532:486]: Mifare classic
[08:08:10][E][nfc:058]: Error, Can't decode message length.
[08:08:10][D][LockEntity:277]: HK SUPPORTED VERSIONS: 00.5C.04.02.00.01.00.90.00 (9)
[08:08:10][D][esp-idf:000]: I (517955) HKAuthCtx: HKAuthenticationContext > Initialization Time: 78 ms

[08:08:11][D][LockEntity:269]: 86.41.04.75.A1.23.05.39.7D.6C.77.B4.18.DB.B9.29.44.C0.06.3A.C9.18.80.F3.5D.A9.20.9D.33.E1.D1.F5.6E.22.78.C5.F5.8E.84.6C.2F.5F.B9.FA.7B.93.CE.B1.65.A7.33.52.62.9B.37.61.D6.4F.42.A2.79.1F.CE.21.BC.EF.8C.9D.10.F8.70.70.52.65.FF.72.D7.C5.E2.22.40.66.54.69.68.90.00 (87)
[08:08:11][D][LockEntity:269]: 90.00
[08:08:11][D][esp-idf:000]: I (518187) HKAuthCtx: authenticate > Endpoint authenticated, transaction took 229 ms

[08:08:11][I][TEST:063]: IssuerID: 8a31148f590dcfb9
[08:08:11][I][TEST:064]: EndpointID: 5d1b48dbba90
[08:08:11][D][pn532:297]: Found new tag '08-69-D8-AF'
[08:08:11][D][pn532:501]: Waiting to read next tag
[08:08:11][W][component:237]: Component pn532 took a long time for an operation (410 ms).
[08:08:11][W][component:238]: Components should block for at most 30 ms.
[08:08:11][D][pn532:486]: Mifare classic
[08:08:11][E][nfc:058]: Error, Can't decode message length.
[08:08:11][D][LockEntity:277]: HK SUPPORTED VERSIONS: 00.5C.04.02.00.01.00.90.00 (9)
[08:08:11][D][esp-idf:000]: I (518564) HKAuthCtx: HKAuthenticationContext > Initialization Time: 77 ms

[08:08:11][D][LockEntity:269]: 86.41.04.E9.CC.B6.7D.D0.56.2A.BD.EE.C2.86.2E.F5.7E.16.64.2E.D4.F4.83.A5.E6.03.24.4C.69.FC.B3.57.70.32.F5.07.66.43.CA.F0.FC.B8.E4.2E.44.45.9A.40.85.13.62.F7.43.E8.CA.87.5E.A8.64.29.A3.05.C9.D7.FC.73.3E.9D.10.A2.EB.73.40.65.D2.95.17.44.66.42.63.00.AA.1C.0B.90.00 (87)
[08:08:11][D][LockEntity:269]: 90.00
[08:08:11][D][esp-idf:000]: I (518789) HKAuthCtx: authenticate > Endpoint authenticated, transaction took 221 ms

[08:08:11][I][TEST:063]: IssuerID: 8a31148f590dcfb9
[08:08:11][I][TEST:064]: EndpointID: 5d1b48dbba90
[08:08:11][D][pn532:297]: Found new tag '08-CF-97-2E'
[08:08:11][D][pn532:501]: Waiting to read next tag
[08:08:11][W][component:237]: Component pn532 took a long time for an operation (399 ms).
[08:08:11][W][component:238]: Components should block for at most 30 ms.
[08:08:12][D][pn532:486]: Mifare classic
[08:08:12][E][nfc:058]: Error, Can't decode message length.
[08:08:12][D][LockEntity:277]: HK SUPPORTED VERSIONS: 00.5C.04.02.00.01.00.90.00 (9)
[08:08:12][D][esp-idf:000]: I (519159) HKAuthCtx: HKAuthenticationContext > Initialization Time: 77 ms

[08:08:12][D][LockEntity:269]: 
[08:08:12][D][esp-idf:000]: E (519380) HKAuthCtx: authenticate > Response not valid, something went wrong!

[08:08:12][I][TEST:067]: IT FAILED
[08:08:12][D][pn532:297]: Found new tag '08-22-7B-B8'
[08:08:12][D][pn532:501]: Waiting to read next tag
[08:08:12][W][component:237]: Component pn532 took a long time for an operation (392 ms).
[08:08:12][W][component:238]: Components should block for at most 30 ms.
```

What am I doing wrong or what am I missing? Do I maybe first have to tell the tool that the scanned Apple Watch / iPhone tag is allowed? If so, how do I do that? As it seems that compared to the component HomeKit-ESP32, there is no web server running on the device that allows to do any configuration?

paskovitch, 2024-08-23:
I have the same problem. Is there a solution for it yet?

rednblkx, 2024-08-23:
Swear to god i thought i had replied to this, sorry about this  :sweat_smile: , got distracted and forgot about it.

For short:
- Don't worry about `Can't decode message length.`, have also now updated the page to reflect this
- Only `on_hk_success` or `on_hk_fail` gets called at the end of the homekey flow, which means your template would not be actioned automatically

If you want to action `cupboard_lock` on hk, you'll need to add it to `on_hk_success`, something like `id(cupboard_lock).unlock();` within the lambda or `- lock.unlock: cupboard_lock` in place of the lambda, see esphome [docs](https://esphome.io/components/lock/index.html), i have updated the example yaml to show this aspect.

paskovitch, 2024-08-24:
Thank you, it works now!

the-smart-home-maker, 2024-08-24:
Awesome, thank you so much. Didn't think it was that easy 😎 👍🏻 we can close this ticket then

closed 2024-08-24 by the-smart-home-maker