rednblkx/HAP-ESPHome #14: Time interval between two successful authentications

paskovitch, 2024-08-24:
After successful authentication, the lock changes its state multiple times if I leave the phone in front of the RFID reader. There should be a parameter between two RFID authentications to define the minimum time interval between two reads. Is this achievable with an ESPHome parameter?
Thank you!

rednblkx, 2024-08-26:
Hi,

Right now there's no delay parameter implemented for such use but it's a work in progress, however, it's a bit tricky due to the way express mode functions and the structure of the pn532 component, which is why an inter-scan delay is not yet present.

rednblkx, 2024-12-07:
Hi,

I've implemented another solution, it should now just wait until it no longer detects anything in the field, please give it a try.

Thanks!

paskovitch, 2024-12-11:
Hi,
With this modification of yours, it works well! Thank you very much! 👍

closed 2024-12-11 by rednblkx