rednblkx/HAP-ESPHome #20: Read traditional NFC tags as well

carlos-leopoly, 2024-11-11:
Could you expose to lambda the data of a traditional tags as well? Like Endpoint ID and Issuer ID.
It would be nice to also read/authenticate with tradtional NFC cards.

I could use to some additional attributes like:
- type_of_card:
  - possible values: "homekey" | "NFC"
- tag_id:
  - example: "EB-37-B5-02"
- NDEF formatted records:
  - example: {"U": "https://open.spotify.com/playlist/0XFQ....."
                     "T": "Stranger Synths"
                    }

Thanks!

carlos-leopoly, 2024-11-12:
Hi rednblkx,
Thanks for paying attention to my request. I'm quite new to ESPHome so probably that's why I have not considered your suggestion to simply use the pn532 component in a standalone way. I would like to combine your project with the [tagreader project of adonno](https://github.com/adonno/tagreader). So probably I'll just copy the config entries from that project.

So you should be right, that's out of the scope of your project. Thank you anyway and thank you for your current project as well!

rednblkx, 2024-11-12:
Alright, great! Let me know if you have any issues.

closed 2024-11-12 by rednblkx

ss83, 2024-11-23:
Hi,
first of all thank you for your work! This project is great, especially the way I can now use Homekey for my automations.

I had the same idea and wanted to use the ESPHome pn532 binary_sensor for normal NFC Tags.

The ESPHome code looks like:
```
binary_sensor:
  - platform: pn532
    uid: 12-34-56-78
    name: "PN532 NFC Tag"
```

When compiling it fails with the error:
```
binary_sensor.pn532: [source /config/homekey.yaml:82]
  
  Platform not found: 'binary_sensor.pn532'.
```

In a separate device without the HAP-ESPHome component it works fine.
I am not a programmer, so this is the only input I can provide for now.

Best regards

carlos-leopoly, 2024-11-23:
Hi @ss83,

This approach does work for me:
```
pn532_spi:
  id: pn532_board
  cs_pin: 5
  update_interval: 100ms

  on_tag:
  - ...
```

Jbcbro, 2024-12-12:
Did anyone get it working ?

carlos-leopoly, 2024-12-15:
Yes it’s working for me. Please take a look at my previous comment