rednblkx/HAP-ESPHome #25: Android App

jappyjan, 2024-12-14:
Hello,

i would like to know if you think it would be possible to create an android app to emulate Homekit behaviour so that android phones can be used (in a secure way) to unlock this DIY ESPHome lock too.
Im pretty new to NFC/Homekey development, therefore i dont really know how to start a project like this :D 
If you could point me in the right directions i guess i could try my best to work it out myself.
Or is there maybe already an android standard for homekey-like functionality? if so could we integrate it?

Kind regards
Jan

rednblkx, 2024-12-14:
Hi there!

In short, no, there is no possibility of using HomeKey on an Android phone, mainly due to cryptographic constraints. 

As to emulating HomeKit, to my knowledge there is no project doing that and it's probably because it wouldn't make much sense, even so now that we have Matter but either way wouldn't help Homekey as HomeKit is only a vessel.

However, there is an upcoming Matter standard called Aliro that's supposed to release in 2025, most likely Q1, and it will bring Homekey functionality to both iOS and Android but some early insights are indicated that uncertified devices(like this) will lack Express Mode.

Let me know if you any other questions.

jappyjan, 2024-12-14:
those are great news! ill dive deeper into this matter feature, thank you a lot!

closed 2025-01-02 by rednblkx