rednblkx/HAP-ESPHome #29: Identify who unlocked

vitinhosessa, 2025-01-11:
I'd like to identify who opened the door so I can make different automations depending on who unlocked the door.

chbmuc, 2025-01-26:
I don't think you can identifiy the person. However each device has it's own `EndpointID`. You can use this ID to distinguish between the devices (once you have the IDs). The code should look something like this:

```
      on_hk_success:
        lambda: |-
          ESP_LOGI("HEREHERE", "EndpointID: %s", y.c_str());
          if (y == "0123456789abcd") {
            id(my_switch).turn_on();
          } else {
            ESP_LOGI("HEREHERE", "Invalid Endpoint");
          }

rednblkx, 2025-01-26:
you can identify a person, as i've explained on my other repo, the `IssuerID` is unique to the Apple ID and the `EndpointID` is unique for each device.

Given that you could use like this:

```
       on_hk_success:
        lambda: |-
          ESP_LOGI("HEREHERE", "IssuerID: %s", x.c_str());
          if (y == "0123456789abcd") {
            id(my_switch).turn_on();
          } else {
            ESP_LOGI("HEREHERE", "Invalid Issuer");
          }
```

I'll see to update the page to make it more clear