rednblkx/HAP-ESPHome #31: Not able to compile

kovkus, 2025-01-30:
**Describe the bug**
With following config I'm getting compile error:

`esphome:
  name: homekeyreader
  friendly_name: HomeKeyReader
  min_version: 2024.11.0
  name_add_mac_suffix: false

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    version: 5.2.1
    platform_version: 6.7.0
    sdkconfig_options:
      CONFIG_COMPILER_OPTIMIZATION_SIZE: y
      CONFIG_LWIP_MAX_SOCKETS: "16"
      CONFIG_MBEDTLS_HKDF_C: y

      
external_components:
  - source: github://rednblkx/HAP-ESPHome@main

# Enable logging
logger:

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
- platform: esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  ap:
    ssid: "HomeKeyReaderHotspot"
    password: "5ibhH5pREWqq"

captive_portal:

homekit_base:
  setup_code: '466-37-726'

    
homekit:
  lock:
    - id: this_lock
      meta:
        manufacturer: "AMICI&CO"
        model: "IMPEDIO"
        serial_number: "42424242"
        fw_rev: "0.16.2"
      on_hk_success:
        lambda: |-
          ESP_LOGI("HEREHERE", "IssuerID: %s", x.c_str());
          ESP_LOGI("HEREHERE", "EndpointID: %s", y.c_str());
      on_hk_fail:
        lambda: |-
          ESP_LOGI("GSDGSGS", "IT FAILED :(");
      hk_hw_finish: "GOLD"

lock:
  - platform: template
    id: this_lock
    name: "HomeKit Lock"
    optimistic: True
[... 129 lines cut ...]
src/esphome/components/homekit/lock.cpp: In member function 'void esphome::homekit::LockEntity::setup()':
src/esphome/components/homekit/lock.cpp:319:29: error: 'accessory_info' was not declared in this scope
  319 |             .model = strdup(accessory_info[MODEL]),
      |                             ^~~~~~~~~~~~~~
src/esphome/components/homekit/lock.cpp:319:44: error: 'MODEL' was not declared in this scope
  319 |             .model = strdup(accessory_info[MODEL]),
      |                                            ^~~~~
src/esphome/components/homekit/lock.cpp:320:51: error: 'MANUFACTURER' was not declared in this scope
  320 |             .manufacturer = strdup(accessory_info[MANUFACTURER]),
      |                                                   ^~~~~~~~~~~~
src/esphome/components/homekit/lock.cpp:321:45: error: 'FW_REV' was not declared in this scope
  321 |             .fw_rev = strdup(accessory_info[FW_REV]),
      |                                             ^~~~~~
src/esphome/components/homekit/lock.cpp:323:26: error: 'hkFinishTlvData' was not declared in this scope
  323 |             .hw_finish = hkFinishTlvData.get(),
      |                          ^~~~~~~~~~~~~~~
src/esphome/components/homekit/lock.cpp:331:28: error: 'NAME' was not declared in this scope
  331 |         if (accessory_info[NAME] == NULL) {
      |                            ^~~~
src/esphome/components/homekit/lock.cpp:337:28: error: 'SN' was not declared in this scope
  337 |         if (accessory_info[SN] == NULL) {
      |                            ^~
*** [.pioenvs/homekeyreader/src/esphome/components/homekit/HAPAccessory.cpp.o] Error 1
*** [.pioenvs/homekeyreader/src/esphome/components/homekit/lock.cpp.o] Error 1
========================== [FAILED] Took 5.55 seconds ==========================`

NonaSuomy, 2025-03-25:
Try changing to these:
    version: 5.3.1
    platform_version: 6.8.1

NonaSuomy, 2025-03-27:
Looks like you have to have the pn523 defined even if you are not going to use it.

closed 2025-08-14 by rednblkx