rednblkx/HAP-ESPHome #45: Not able to add accessory to HomeKit

nickwilkie, 2025-08-22:
I've got HAP-ESPHome installed and running with a single lock entity. Everything is connected, and it's recognized when I look for accessories to add. When I try to add it, my phone says "Unable to Add Accessory, This acessory cannot be used with HomeKit". I've tried playing around with the homekit_base parameters, but not able to get connected. I'm on the same wifi network, no client isolation.

In the ESPHome logs, I get:
```
[13:28:43][D][LockEntity:074][httpd]: hap_event_handler > Found allocated controller - Hash: 
[13:28:43][D][LockEntity:079][httpd]: hap_event_handler > Issuer  already added, skipping
```

My config is:

```
substitutions:
  name: box-s3
  friendly_name: BOX-S3

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2025.5.0
  name_add_mac_suffix: false

esp32:
  board: esp32s3box
  flash_size: 16MB
  cpu_frequency: 240MHz
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240: "y"
      CONFIG_ESP32S3_DATA_CACHE_64KB: "y"
      CONFIG_ESP32S3_DATA_CACHE_LINE_64B: "y"

      CONFIG_COMPILER_OPTIMIZATION_SIZE: y
      CONFIG_LWIP_MAX_SOCKETS: "16"
      CONFIG_MBEDTLS_HKDF_C: y

external_components:
  source: github://rednblkx/HAP-ESPHome@main
  refresh: 0s

lock:
  - platform: template
    id: "s3_lock"
    name: "S3 Box Lock"
    optimistic: True
    on_lock:
    - logger.log: "Door Locked!"
    on_unlock:
    - logger.log: "Door Unlocked!"

homekit:
  lock:
    - id: s3_lock
      nfc_id: nfc_spi_module
      on_hk_success:
        lambda: |-
          ESP_LOGI("HEREHERE", "IssuerID: %s", x.c_str());
          ESP_LOGI("HEREHERE", "EndpointID: %s", y.c_str());
      on_hk_fail:
        lambda: |-
          ESP_LOGI("GSDGSGS", "IT FAILED :(");
      hk_hw_finish: "SILVER"

homekit_base:
  meta:
    name: "PRIMO"
    manufacturer: "AMICI&CO"
    model: "IMPERIUM"
    serial_number: "16161616"
    fw_rev: "0.16.2"
  setup_code: '159-35-728'
  setup_id: "ES32"

button:
- platform: homekit_base
  factory_reset:
    name: "Reset HomeKit pairings"

spi:
  clk_pin: 38
  miso_pin: 39
  mosi_pin: 40

pn532_spi:
  id: nfc_spi_module
  cs_pin: 41
  update_interval: 100ms

logger:
  level: DEBUG
  hardware_uart: USB_SERIAL_JTAG

api:
  encryption:
    key: "**********"

ota:
  - platform: esphome
    id: ota_esphome
    password: "*****************"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: "Box-S3 Fallback Hotspot"
    password: "*******"
```

rednblkx, 2025-09-02:
Could be a glitch, probably things got confused for a moment, there is currently a bug where it’s not able to connect in some conditions but I cannot reproduce the issue you are seeing.

I’ll update this issue if anything comes up.

segdy, 2025-09-13:
But you were able to find it at least?
It's driving me nuts

[image]

It showed up two times, randomly, without having done anything specific (and then I got same error as above). But now it doesn't even find it any more. I tried everything I could think of:

1. Two separate phones
2. Added home hub
3. Phone with and without internet access
4. ESP with and without internet access
5. Same L2 network
6. Various yaml changes
7. Everytime a complete new re-flash
8. 15W wall charger
9. Removing lock and just using a simple "Test Sensor"

philippwaller, 2025-10-26:
Hey @eigenphase & @nickwilkie,

quick heads-up: after I tweaked my config and reflashed today, I could no longer add the bridge to HomeKit. The error was:

> Unable to Add Accessory, This acessory cannot be used with HomeKit

I recovered it like this:

1. Fully erase the flash (macOS/Linux; find your port with `ls -al /dev/` and update the command):

```
esptool --port /dev/cu.usbserial-1320 erase-flash
```
2.  Reflash and start with the default config below before changing anything.
Please don’t tweak settings initially—the defaults are (apparently) very sensitive and even small changes prevented pairing for me. 

**However, do adjust the GPIOs to match your wiring and board!**

```
esphome:
    name: homekey-test

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

lock:
  - platform: template
    id: "this_lock"
    name: "Main Lock"
    optimistic: True
    on_lock:
    - logger.log: "Door Locked!"
    on_unlock:
    - logger.log: "Door Unlocked!"

external_components:
  source: github://rednblkx/HAP-ESPHome@main

spi:
  clk_pin: 14
  miso_pin: 12
  mosi_pin: 13

pn532_spi:
  id: nfc_spi_module
  cs_pin: 5
  update_interval: 100ms

esp32:
  board: esp32dev
  flash_size: 4MB
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_COMPILER_OPTIMIZATION_SIZE: y
      CONFIG_LWIP_MAX_SOCKETS: "16"
      CONFIG_MBEDTLS_HKDF_C: y

homekit_base:
  setup_code: '159-35-728'

homekit:
  lock:
    - id: this_lock
      nfc_id: nfc_spi_module
      on_hk_success:
        lambda: |-
          ESP_LOGI("HEREHERE", "IssuerID: %s", x.c_str());
          ESP_LOGI("HEREHERE", "EndpointID: %s", y.c_str());
          if (id(this_lock).state == LOCK_STATE_LOCKED) {
              id(this_lock).unlock();
          } else {
              id(this_lock).lock();
          }
      on_hk_fail:
        lambda: |-
          ESP_LOGI("GSDGSGS", "IT FAILED :(");
      hk_hw_finish: "SILVER"

logger:
  level: DEBUG
```

If this pairs successfully, make changes one at a time and re-test pairing after each change.