rednblkx/HAP-ESPHome #50: Exposing a device without a bridge?

eigenphase, 2025-09-18:
Something I recently learned is that HomeKit supports ordinary devices and hubs (that can expose multiple devices). The Home Assistant HomeBridge extension actually let's me choose if I want to expose a single device or multiple devices plus Hub.

The initial project (https://github.com/rednblkx/HomeKey-ESP32) seems to expose a single lock device, not a hub.

This project however, seems to expose a hub with one device.

Is it possible to just omit the hub and expose a lock directly? 
If not, is there a specific reason?

rednblkx, 2025-09-18:
Currently that is not an option because of the architecture.
When I created the project it was just easier to implement it by just creating a bridge accessory that links all sub-accessories. I might implement this in the future but since it doesn’t make an actual difference it’s not an important change.

rednblkx, 2025-09-19:
Because those two projects serve two distinct purposes, HAP-ESPHome is meant to expose multiple different types of entities not just a “lock” while HomeKey-ESP32 is only meant to provide one thing and that is a single virtual lock with the HomeKey functionality.